tschroedter/idasen-desk #1: How can use this one?

zeroarst, 2020-11-29:
I am using Windows 10, and I have tried other github repos but none work for me...is this one working for windows?

tschroedter, 2020-11-29:
Hi, 
Yes, the code in this repo is based on Windows 10 and .Net Core 3.1.
Let me know if you need more information.

zeroarst, 2020-11-29:
How can I use it..? download the repo and open with IDE like visual studio? I am just wondering if this will be packed to an app?

tschroedter, 2020-11-29:
1. Correct, download the repo.
2. Double click the Idasen-Desk.sln file to open. (This will open Microsoft Visual Studio) 
3. Build the solution by selecting Build->Build Solution.
4. Click the greem Play button to start the console application.

Notes:
- At the moment this is just a simple console app to move the desk to a given height. You can change the height in the Demo.cs file. Line 126, change the value in "_desk.MoveTo ( 7200u ) ;"  7200 = 72cm

- Next thing I'll try to write a little app which will allow to preset heights and move to them.
- After that a little WPF application which hides into the system tray.

zeroarst, 2020-12-12:
I am trying to open the solution project, so I installed Visual Studio 2019. 
But when I opened it it says incompatible projects. Maybe my VS is too new??
[image]
[image]

closed 2020-12-12 by zeroarst

tschroedter, 2020-12-13:
Just to let you know the latest repository contains a system tray application which allows you to control the desk. It's still an  early version but the basic functionality works. Let me know if you are still interested and I can tell you how to run it.

zeroarst, 2020-12-13:
Awesome. I just pull the latest version and it works!! 

Couple days ago I just installed VS 2019 and solved some issues because I have not yet used Visual Studio for over 10 years. But was not still working after I built it. 

Today it works after I just pulled the latest version. It would be good to see more functions such as presets. 

I don't see any any relevant application in system tray, should it appear after I run it?

zeroarst, 2020-12-13:
@tschroedter 
Yes please. I installed Visual Studio 2019 and solved some problems, such as I need to install C# development. But finally got it working.
I can only see a command line window pops up and many logs.
[image]

Please shared with me how to get the application shows in the system tray.

tschroedter, 2020-12-13:
Yes, you should see a little blue cup icon in the system tray area. Look at the README.md which is shown on the Github site:
https://github.com/tschroedter/idasen-desk

The system tray application is not running If you can't see the blue cup icon in the system tray area.

Steps to get it running in VS 2019:
- Get the repository _(you've done that)_
- Open the _Idasen-Desk.sln_ file in VS 2019 _(you've done that)_
- Open the _Solution Explorer_ in VS 2019 by clicking _View->Solution Explorer_ from the menu at the top of VS 2019.
- In the Solution Explorer windows select the _Idasen.SystemTray_ project.
- Right click on the _Idasen.SystemTray_ project and select Set as _Startup Project_ from the context menu.
- Build and run the project...
-- ...by pressing the F5 key on your keyboard.
-- ... or press the little green play button below the main menu

Let me know if this works.

PS: I'm currently looking to build an installer or provide access to a single .exe file.

tschroedter, 2020-12-13:
Your last message just poped up. - I think it's running the console application by default which is just for testing.

You have to tell VS 2019 to run _Idasen.SystemTray_ as described in my last comment.

zeroarst, 2020-12-13:
Yes. It works now. It is really awesome!! Standing and Seating are working. Really nice. Thank you for the work! 
[image]

Do you know how to make it as an executable application so I can let it run when Windows starts up? Or maybe in the future you will provide an option to get it run on startup?

By the way, I just ran into an exception but I don't know how to reproduce it. Maybe I will keep playing with it to find out.

Again, thank you for the work!

tschroedter, 2020-12-13:
In regards to single file exe:
- Open the solution in VS 2019
- Click _Build-> Pulish Idasen.SystemTray_ from the top menu.
- Note the _Target location_ folder, this folder will contain the exe file.
- Click the _Publish_ button.
- Let it do its thing...
- Open the _Target location_ folder in the Windows Explorer by double clicking the _blue Target location folder_.
-- There you have your single _Idasen.SystemTray.exe_

In regards to running it on startup:
- Create a shortcut on the desktop by right clicking _Idasen.SystemTray.exe_
- Goto the normal Windows Desktop and copy the shortcut by right clicking it and select copy
- Follow the steps from Microsoft [Add an app to run automatically at startup in Windows 10](https://support.microsoft.com/en-us/windows/add-an-app-to-run-automatically-at-startup-in-windows-10-150da165-dcd9-7230-517b-cf3c295d89dd)

Hope set helps.

In regards to the exception, can you create a new issue and copy the logfile or exception into it. Thanks!

I'm pretty sure there are more issues because you are using the first version ever :)