tschroedter/idasen-desk #2: Power cycle needed if hardware button is used during motion

lukabelingar, 2022-01-19:
I pressed the on-table button after move command was issued from software.
Software falls into somekind of dead loop, rapidly displaying cannot connect message, table accepts no further commands, not even from the hardware on-table button until power cycled. Clean Exit from program was difficult due to popup, but possible using navigational keys.

tschroedter, 2022-01-21:
Hi, I'll try to replicated it.

tschroedter, 2022-01-22:
Hi, I was able to reproduce it. Basically the manual override stops the movement of the desk but the app doesn't know about it. I still thinks it's moving the desk but the returned desk speed is zero. When I have more time I'll have a closer look.

Work around:

- Use the _Disconnect_ and _Connect_ menu option in the context menu.
- Don't push the manual button :)

tschroedter, 2022-01-24:
I think I fixed it in the latest version 90. Can you try again?

lukabelingar, 2022-01-25:
Doesn't work; though I think you're up to a very cool parental feature.
It worked exactly once for me - after I stopped the desk by leaning on it, app showed desk height notification and the it was possible to move it further. The rest of the time - either using gyro or button to stop movement and after app restart, result was the same - no control was possible, either from app or from physical button.
Which brings me to the aforementioned cool parental feature - freezing the physical switch. Is that something your app could be modified to cause intentionally?

tschroedter, 2022-01-26:
I added the hidden parental feature to see if someone finds it. :) - No, sorry just an unexpected side effect. I'm not sure why it is doing it on your machine/desk. I wasn't able to reproduce the freezing here. Maybe, the physical desk controller got conflicting inputs and decided to give up. :)

I think I might have found and fixed why it only worked once. When you have time give it another go with V0.0.94

tschroedter, 2022-01-26:
I think the issue is fixed. Let's see if someone else gets the same issue.

closed 2022-01-26 by tschroedter

lukabelingar, 2022-01-27:
why do you keep calling it "issue"? i was sincerely happy with the behavior and am reluctant to let go of it :)
did you at least note in code comments, when did you change it? or if you can describe cause in a few words? i might implement it properly myself.

lukabelingar, 2022-01-27:
BTW, yes, "issue" fixed.

tschroedter, 2022-01-28:
I'm not 100% sure because I couldn't regrade the 'parental feature'. My best guess is that the app continues to send desk movement commands which are ignore because the physical up/down button was pressed.  The app doesn't know that the desk isn't really moving. The log shows current speed 0 but it continues to send the move up or down command because it wants to get to the target height. This was fixed by keeping a history of the last few movements see DeskMover and DeskHeightMonitor. 

If I would want to implement the 'parental feature' then I would create a new class DeskParentalControl which would listen to the height changes which you can get from DeskHeightAndSpeed. Everytime you detect a height change not coming from the app you send a stop command to the desk (DeskMover). Let me know if you want to have a go and I can create a branch and provide you with skeleton classes which you can populate with code.

lukabelingar, 2022-01-28:
Tempting :) Though, I'm in a pickle - in order to have more free time, my kids would have to be all grown up, but what would be the point then... So, thanks for your kindness and updates, but for now I think I'll stick with scolding.