tschroedter/idasen-desk #3: Desk not detected if its name does not start with Desk

lukabelingar, 2022-01-19:
I decided to personalize my desk using official sw. This sw did not detect the desk, because it only checks for Desk* BT devices. This could be a setting along the sitting/standing height.

tschroedter, 2022-01-21:
Hi, 

I'll have a look on the weekend and most likely add a new setting for desk name and desk address. At the moment it's hard coded and detects a desk by the name starting with 'Desk' or the address.

lukabelingar, 2022-01-21:
That's great! Yes, I saw it in the code but am fairly new to .net and did not want to dig in right away :)

tschroedter, 2022-01-22:
I just released a new version which allows you to specify the desk name or desk address in the settings.

Can you try it out and let me know if it works for you?

lukabelingar, 2022-01-22:
It works; though with some issues:
1. I had to uninstall the program first in order to see the new UI elements. Simply installing the new version while old one was stopped was not enough.
2. I had to re-start the program after setting my new desk's name. StartsWith() works, though I would prefer two ways of finding desk - automatic with your "Desk<space>NNNN" pattern if manual setting was empty, or else name would have to match, caseless.

tschroedter, 2022-01-24:
In regards to 1, not sure how to fix it yet. It's a Windows Installer thing. Sorry.
In regards to 2, the new version 90 will automatically disconnect and reconnect to the desk when you change the Advanced settings. Can you try and let me know if it works for you?

lukabelingar, 2022-01-24:
Um, yeah, about 1. ... why did you even go with the installer and not just a "portable app" aka exe? installer is unsigned, so security warning is there in any case. 2. - will check tomorrow.

lukabelingar, 2022-01-25:
2. no, does not work for me. Though after changing desk's name windows kept connecting to the old name (i guess via its address). Steps: turn off BT on PC, connect to table with Android, change name, delete BT device on PC, turn on BT on PC, add new BT device, change app settings. App was somehow connected, because for a brief moment (0.2 sec or so) table moved toward commanded position.

tschroedter, 2022-01-26:
1. I went for the installer because I could easily make the app start-up when Windows starts. The next thing I do won't use installers. :)
2. I think I found/fixed the reconnect but I will try again following your steps. SO far I was to lazy to change the desk name.

tschroedter, 2022-01-26:
I close the issue. Looks like a WIndows issue when you change the device name after it was already paired.

closed 2022-01-26 by tschroedter