tschroedter/idasen-desk #8: Change settings of table height and possible hotkeys

ducklingasa, 2022-03-28:
Hi Tom,

Thank you for this great app!
Would it be possible to extend the height in the settings from 60cm to 127cm (the full range of the table)?

Also if you could add global hotkeys (e.g. CTRL+ALT+1 and 2) to move into seating and standing positions would be great :)

Cheers,
Adam

ducklingasa, 2022-03-30:
Hi Thomas,

1. It worked perfectly, thanks for your time.
2. I've found some example codes here: [Keyboard Hook in C#](https://docs.microsoft.com/en-us/archive/blogs/toub/low-level-keyboard-hook-in-c)

Regards,
Adam

tschroedter, 2022-04-04:
Hi Adam,

Thanks for the link to the Keyboard Hooks I will try it in the next few days. I also found a few NuGet packages doing simalar things. I think I have to find the solution which doesn't trigger th virus scanner.

Regards,
Thomas

tschroedter, 2022-04-18:
Hi Adam,

I published a new version 0.0.111 which includes support for hot keys. Can you have a quick look?

Reagrds,
Thomas

ducklingasa, 2022-04-18:
Hi Thomas,

I have tested it just now and it works perfectly. Great job, thanks!

Regards,
Adam

tschroedter, 2022-04-23:
No worries. Thanks for testing it.

closed 2022-04-23 by tschroedter