tschroedter/idasen-desk #9: Split the application into a windows service and UI

7orlum, 2022-03-28:
Would it be possible to split the application into a windows service and UI? I see that the windows service will support custom commands that will allow us to use automation programs, remote controls and keyboards to control the height of the desk

tschroedter, 2022-03-29:
Yes, on the develop branch I've got a project which hides the desk control behind a REST API. At the moment it's a console app which can be easily converted into a Windows Service using the NuGet package TopShelf. Let me know if the REST API is 'good' enough and I will create a branch or fork with only the REST API as a service.

7orlum, 2022-03-29:
Yes, it's a good option, thank you, Tom

tschroedter, 2022-04-04:
Hi, I created another repository for the REST API. This is an early version which runs the API as an console app.

https://github.com/tschroedter/idasen-desk-rest-api

tschroedter, 2022-04-09:
I pushed the first working version to the repository. Have a look and let me know what you think.

tschroedter, 2022-04-23:
The REST API is working [idasen-desk-rest-api](https://github.com/tschroedter/idasen-desk-rest-api) as a console app or Windows Service.

Besides that I pulled out all the Idasen/Bluetooth code into it's own repo: [idasen-desk-core](https://github.com/tschroedter/idasen-desk-core)

closed 2022-04-23 by tschroedter