tschroedter/idasen-desk #13: DllNotFoundException

dc914337, 2022-09-02:
Hey, couldn't run the app. Here is the error from Event Viewer:
`Application: Idasen.SystemTray.Win11(2).exe
CoreCLR Version: 6.0.522.21309
.NET Version: 6.0.5
Description: The process was terminated due to an unhandled exception.
Exception Info: System.DllNotFoundException: Dll was not found.
   at MS.Internal.WindowsBase.NativeMethodsSetLastError.SetWindowLongPtrWndProc(HandleRef hWnd, Int32 nIndex, WndProc dwNewLong)
   at MS.Win32.UnsafeNativeMethods.CriticalSetWindowLong(HandleRef hWnd, Int32 nIndex, WndProc dwNewLong)
   at MS.Win32.HwndSubclass.HookWindowProc(IntPtr hwnd, WndProc newWndProc, IntPtr oldWndProc)
   at MS.Win32.HwndSubclass.SubclassWndProc(IntPtr hwnd, Int32 msg, IntPtr wParam, IntPtr lParam)
`

Desk is paired and called "Desk 023". The app just doesn't run

dc914337, 2022-09-02:
[V0.0.111](https://github.com/tschroedter/idasen-desk/releases/tag/V0.0.111) works for me by the way

tschroedter, 2022-09-03:
Hi,

I can confirm that I get the same issue on Windows 11 (Version 10.0.22000 Build 22000). I'll try to fix it.

Q: Which Windows version do you use 10 or 11?

Regards,
Thomas

dc914337, 2022-09-03:
> Windows 11 Pro. Version: 21H2
> Build: 22000.856

tschroedter, 2022-09-03:
Thanks, I found the issue and fixed it for Windows 10 and 11.

Can you try V0.0.1.71? 
(https://github.com/tschroedter/idasen-desk/releases/download/V0.0.171/Idasen.SystemTray.Win11.exe)

tschroedter, 2022-11-14:
Fixed in V0.0.171

closed 2022-11-14 by tschroedter