tschroedter/idasen-desk #15: update UI

sIRwa2, 2023-02-23:
[image]
May i suggest an UI update? love to help.

emiltorp, 2023-02-23:
I made a fork a while back where I went the other way, instead of all custom design I went full Windows 11 native, with native notifications as well. 

In that fork i also removed all unnecessary packages and Jetbrains stuff, so I never sent a pull request. (Too cluttered)

So even though I like this design, if there should be a design update, consider to go native. At least for the notifications.

zeroarst, 2023-02-23:
> [image] May i suggest an UI update? love to help.

That looks great... would love to see the new UI..

tschroedter, 2023-02-26:
Hi, the suggested UI good. Feel free to fork the repository and implement it. I didn't  have time during the last few month to work on the project. My plan for the near future is to fix the existing bug related to the log file size, add code to support multiple users and update the UI using MAUI. MAUI will give you the native look and feel for Windows 10 and 11.

falkenhawk, 2023-08-18:
@tschroedter is it possible to disable all notifications altogether, e.g. via `appsettings.json` file?

tschroedter, 2023-08-18:
This feature doesn't exist yet. I'll have a look today or tomorrow and see if I can quickly add it. It shouldn't be to hard to do.

tschroedter, 2023-08-20:
@falkenhawk I added a new advanced setting to show/hide all notifications (see [V0.0.189.0](https://github.com/tschroedter/idasen-desk/releases/tag/V0.0.189). Let me know what you think.

falkenhawk, 2023-08-21:
Does exactly what it is meant to :) Thank you!

tschroedter, 2024-10-22:
Done.

closed 2024-10-22 by tschroedter