tschroedter/idasen-desk #16: Can't determine desk height

AlexandreLogist, 2024-10-28:
For some reason I am not able to connect to my desk (I can when using the Linkal app).
(Connecting pressing 3s the button till the blue led flashes.

Log does not contain a lot of info I'm afraid. Any idea?

[image]
[Idasen.SystemTray.log](https://github.com/user-attachments/files/17548733/Idasen.SystemTray.log)

tschroedter, 2024-10-28:
Hi,

### Quick Fix
First, can you try to move the desk manual and see if the desk height is updated?
- The desk doesn't provide the current height. The desk has to be moved a tiny bit get the height.

Second, are you sure the desk is paired/connected in Windows?
- Bluetooth is on
[image]
- Desk is connected 
(Correct you have to press the button till the blue led flashes)
[image]

### How to connect the desk?
- Enable BlueTooth
[image]
- Click 'More Bluetooth settings'
[image]
- Click 'Add device' in the 'Bluetooth & devices' window
[image]
- Click 'Bluetooth' in the 'Add a device' window
[image]
- Click the button on the desk until the blue light flashes and the desk should appear in the 'Add a device' window
- Click 'Desk...' to pair/connect to the desk
[image]
- Click 'Done' on the 'Your device is ready to go!' window
[image]

Let me know if it works.

Regards,
Thomas

PS: I think I need to have a log at the logging, you are right it doesn't tell you much at the moment.

Evo589, 2024-12-16:
I had problems getting Windows 11 to even see my desk in pairing mode. Turns out you need to set the Bluetooth device discovery to advanced.

https://www.reddit.com/r/cricut/comments/14h9sz8/windows_11_bluetooth_issues_fixed/

tschroedter, 2024-12-17:
Hi Evo589,

Great you found a solution and letting me know! - I never had the problem, I guess I was lucky :)

tschroedter, 2025-01-21:
Closing the issue.

closed 2025-01-21 by tschroedter