tschroedter/idasen-desk #17: The application closes after any action with the desk

Froze89, 2025-01-05:
Hello,
The application starts and works until I change the height of the table through the application. After the height of the table has changed, the application immediately closes itself and disappears from the tray.

Froze89, 2025-01-05:
One more problem, settings are not saved, height cannot be saved for sitting and standing positions

tschroedter, 2025-01-05:
Hi,

maybe it's a problem with the settings file or permissions on the file.

1. Can you stop the app, delete the settings file and try again?

2. Can you try to change the permission on the settings file? 
(The current user should have modify, read, write permission.)

PS: The location of the settings file can be found under Settings->Advanced->Settings File

Froze89, 2025-01-05:
It's very strange but Windows says that I don't have the settings file on my computer. I tried to restart it, turned off the antivirus but the file still isn't created.

Froze89, 2025-01-05:
From log file

Checking if 'C:\Users\msirt\Desktop\Programs\DeskControl\appsettings.json' exists...
...no, 'C:\Users\msirt\Desktop\Programs\DeskControl\appsettings.json' does not exists.
Using default settings...
Using the following configuration:
Serilog:
  Enrich:
    0=FromLogContext (JsonConfigurationProvider for 'appsettings.json' (Required))
    1=WithMachineName (JsonConfigurationProvider for 'appsettings.json' (Required))
    2=WithProcessId (JsonConfigurationProvider for 'appsettings.json' (Required))
    3=WithThreadId (JsonConfigurationProvider for 'appsettings.json' (Required))

Froze89, 2025-01-06:
I have just fixed this.

I have manually created folder Idasen.SystemTray in the C:\ProgramData\, removed read only in properties of the folder.
Now it works

tschroedter, 2025-01-07:
Thanks for the update. I close the issue because it looks like a permissions issue.

_PS: Maybe the app should show a message if it can't access the settings file but that would be another issue._

closed 2025-01-07 by tschroedter