tschroedter/idasen-desk #19: Add support for additional desk positions (beyond standing/sitting)

merlin334, 2025-09-03:
Hi, 

I'm going to start by saying: This is a great program. I've been using it for about 2 years.

Right now it has standing and sitting which is great, but if you could add more positions would be even greater.
I use it for standing, sitting and 2 more positions, standing on a treadmill and eating at the desk position :).

If is not too hard or time consuming it would be great to have more positions.

Have a nice day.

tschroedter, 2025-09-07:
Thanks for taking the time to prepare the screenshots/mock-ups and thinking about how to make the app better.

In regards to the first screenshot and adding 'Custom 1' and 'Custom 2':

I was able to 'quickly' code it and it seems to be working, including hot keys.

At the moment the 'Settings' looks like this:
[image]

And 'Hot Keys' like this:

[image]

I like your idea about the new layout of the 'Home' page in screenshot 2 but as you said this will take much more time. I was thinking about supporting 2 or more users as well but so far nobody requested it. At the moment the new Custom settings are a bit hacky meaning a better coding solution would be to make the height setting configurable. Meaning you can add as many settings as you want and given them your own names. Adding settings for different users is also possible but also needs some extra time. I think your suggested 2 custom setting is a nice compromise at the moment.

In regards to screenshot 3, I fixed the minimize issue in the latest code already but I haven't released it yet. Can you clarify the functionality behind the checkbox 'If checked show in tray bar' next to the new custom settings (Treadmill/Eating)?

In regards to screenshot 4, your order of the menu items makes sense. 
- Separator => added in the latest code
- Spinner next to the 'Connect' => good idea, not sure if it can be done (Todo)
- Confirmation popup for 'Disconnect' => I can do that, that should be straight forward (Todo)
- 'Show'/'Hide' (depending on if the window is shown) => I think this can be done. (Todo)
- Add icons to each menu item => I think this can be done. (Todo)
- Other things, I need to do some code quality improvements after adding the new settings. (Todo)

Currently the menu looks like this:

[image]

merlin334, 2025-09-07:
Some desks come with 4 preset options (1, 2, 3, 4), so it already looks great!

Your question: Can you clarify the functionality behind the checkbox 'If checked show in tray bar' next to the new custom settings (Treadmill/Eating)?

This checkbox relates to the tray menu (shown in the last screenshot). If it’s checked, the option (e.g., Treadmill) will appear in the tray menu when you right-click the tray icon.

This can be useful if you have a position you don’t use often but want to keep preconfigured—then you can just check/uncheck it (similar to marking a position Active/Inactive) depending on when you need it. For example, some desks already come with 4 presets (1–4), but most people don’t use all of them all the time. Having the ability to hide unused ones makes the tray menu much cleaner.

You could even consider adding a checkbox for the STOP option, so it doesn’t show up in the tray menu if you never use it. Personally, I stop using the manual button on the desk, so I wouldn’t need it in the tray. The fewer unused items you see there, the better.

Currently, double-clicking the tray icon opens the settings. You could also make it so that double-clicking toggles the settings window (hide if already open).

Considering multiple users: This is just a thought. I'm not sure how many people would actually use this feature, so it could be left for later or as an experiment when you have time

tschroedter, 2025-09-18:
Hi,

I created a pre-release including most of you suggestions.

It would be great if you download the new version and let me know what you think.

merlin334, 2025-09-18:
Short version (ignoring the bugs):
The new positions and tray checkboxes look good.
The tray also looks good.
Bluetooth connection seems faster than before.
Showing and hiding settings on clicking the tray icon is great.

Since this is a pre-release, it’s normal to have some bugs. You may already be aware of some of them, but here’s what I encountered:

**Settings**
 - **Navigation**
    When switching to Advanced (or any other tab), scrolling doesn’t work. You have to close that tab to view the other tabs and their content.

- **General tab**
    Positions don’t work as expected:
    Stand at 116 - moves to 109 or maximum
    Sit at 76 - moves to the minimum
    Custom 1 at 85 - jumps to the maximum
    Custom 2 at 124 - jumps to the maximum
    
    Tray checkbox doesn’t apply immediately. It only works after switching to another tab (probably when settings are saved). It would be nicer if it saved right on action - e.g., when clicking the tray checkbox or stopping the height slider drag.
    
    Theme starts in Light even though my OS is set to Dark. Not a big issue since switching it in Appearance works fine.
    
    Stop visibility option feels a bit hidden in Advanced. I expected it under General (maybe near Positions), and only found it later by accident.
    I would rename 'Stop' visibility tray, if it will only affects the tray.
- **Tray**
    Height value bug: after some time, the tray height gets stuck at 109. Restarting the app fixes it.
    Exit button: pressing Exit crashes/force closes the program. It does exit, but in a rough way.

tschroedter, 2025-09-20:
Hi,
Thanks for your time testing the app! Your feedback is really helpful.

In regards, to the bugs...

Navigation
- _"When switching to Advanced (or any other tab), scrolling doesn’t work. You have to close that tab to view the other tabs and their content."_
Fixed. - To get the UI right in WPF is sometimes a real pain. I think I found a solution to get a scrollbar when there isn't enough space. I had to compromise and make the window a fixed size and some other little things.

General tab
- _"Positions don’t work as expected:
Stand at 116 - moves to 109 or maximum
Sit at 76 - moves to the minimum
Custom 1 at 85 - jumps to the maximum
Custom 2 at 124 - jumps to the maximum"_
In Testing - So far I wasn't able to reproduce the problem. I'll try on a different machine when I have time.

- _"Tray checkbox doesn’t apply immediately. It only works after switching to another tab (probably when settings are saved). It would be nicer if it saved right on action - e.g., when clicking the tray checkbox or stopping the height slider drag."_
Fixed. - Correct observation, saving was done only when you switched to a different navigation tab. I changed it so that it saves the changes immediately.  

- _"Theme starts in Light even though my OS is set to Dark. Not a big issue since switching it in Appearance works fine."_
Fixed. - If there are no setting for the theme then we use the OS theme.

- _"Stop visibility option feels a bit hidden in Advanced. I expected it under General (maybe near Positions), and only found it later by accident."_
Fixed. - I moved the 'Stop' visibility checkbox under positions. Not 100% happy but you are right it's a better spot.

- _"I would rename to 'Stop' visibility tray, if it will only affect the tray."_
Fixed. - If you have an idea for a better description, let me know. 

Tray
- _"Height value bug: after some time, the tray height gets stuck at 109. Restarting the app fixes it.
Exit button: pressing Exit crashes/force closes the program. It does exit, but in a rough way."_
Fixed. - Found a hidden stack overflow error in some debugging code which crashed the app randomly.

### New Beta Version
I created new version [V0.0.312](https://github.com/tschroedter/idasen-desk/releases/tag/V0.0.312).

Please, have a look if you have some time.

merlin334, 2025-09-21:
Sorry for the late reply, I wasn’t at my PC over the weekend.

The settings window cannot be resized now and scrolling is not fluent, only jumps from top to bottom.

The positions still don't work as expected:
Stand (116): If it's at 75 and press Stand it moves to 126. If it's at 114, it goes to 118. Pressing Stand again moves it back to 114, and so on.
Sit (76): First it moves to 70. Pressing Sit again moves it to 76, then pressing once more goes to 74.
Custom 1 (85): It moves to 89, then pressing again goes back to 85, then again to 86, and so on.
Custom 2 (124): It moves to 123, then pressing again goes to 125, then back to 123, repeating the cycle.

It does not go to the exact position and if the desk is already at a saved position, pressing that button should not move it at all.
At least on my PC (Windows 11)

tschroedter, 2025-09-25:
Hi merlin334 and Glakt,

v0.0.263 is the stable version at the moment. All the others are beta, I'm trying to add new features, improve the UI and remove some ugly code related with Bluetooth under the hood.

**Regarding the height issue:**
- Yes, I believe I can reproduce it on another machine and I'm investigating it now.

**As for Glakt’s points:**

- 1 & 2. The desk movement stops based on an estimated position in code, which means it usually gets close to, but not exactly at the sit/stand height. The accuracy depends on the current motor speed. Hitting the exact height would require creeping toward it in small increments, which I think would feel clunky. Also, the desk head is only readable while the desk is actively moving.
- 3 Fixed! Smooth scrolling is now enabled in the latest build.
- 4 Resizing the window has been a nightmare. For now, it's fixed-size to keep the UI clean. I plan to revisit resizability later.
- 5 You're right — the sliders are too cramped. I’ll increase the window size to give them more breathing room.

The root of most of these issues is the ongoing refactor of the Bluetooth layer and the code to control the desk movement. WPF (the UI framework) is great when it works, but even tiny missteps can cause big headaches :)

**ToDos:** 
- Check target height toleranz.
- Make window resizable.
- Fix squeezed sliders.

tschroedter, 2025-10-04:
Hi All,

Sorry, I was talking about my local build. The scrolling seems to be fine in the new beta version. It was a real pain to get WPF working correctly and required some extra coding.

**What's changed in the latest beta version:**
- Scrolling is fixed.
- Resizing is fixed.
- Stuttering during desk movement is fixed.
- Hitting the target height is +/- 1 cm at the moment. - Maybe, I can tweak it a bit more.
- Fixed squeezed slider. They will resize with the window.
- Added Up/Down bottoms to change height by 1 cm.
- Change to the desk positions are saved immediately.
- The new Custom 1 and Custom 2 position work as expected.
- Use only one logfile, which is not fully supported by SeriLog so you might see 2 files for a short time.

**New Beta Version:** [Idasen-SystemTray-0.1.77-win-x64.exe](https://github.com/tschroedter/idasen-desk/releases/tag/untagged-9d9cbea897ef3794fc57)

Feel free to download and test it!

_Note: I'm still testing my self and have to run the new beta on another machine._

tschroedter, 2025-10-08:
Hi Glakt,

I created a [Beta Release Page V0.1.177](https://github.com/tschroedter/idasen-desk/releases/tag/V0.1.77).

Or you can download the file directly: [Idasen-SystemTray-0.1.77-win-x64.exe](https://github.com/tschroedter/idasen-desk/releases/download/V0.1.77/Idasen-SystemTray-0.1.77-win-x64.exe)

Hopefully it works this time.